jpmeijers/RN2483-Arduino-Library #1: No valid response from initOTAA

kneibdev, 2016-09-08:
I am using a Sodaq Mbili to try this lib.

the function : initOTAA(String AppEUI, String AppKey);  does register the device (I can see this in the dashboard) but always returns 'false'. So I keep getting the message:

> Unable to join. Are your keys correct, and do you have TTN coverage?

jpmeijers, 2016-09-08:
Hi kneibdev.

I do not have a Sodaq Mbili to test with. Is it similar to the Sodaq One?

If you [listen on MQTT](https://www.thethingsnetwork.org/wiki/Backend/Connect/Application#connect-an-application_streaming-data_topics_activations), do you get a message when the join is performed? If this is the case you are correct that the device is activated. This means that the Join Accept message is not received by your node. This can either be a [bug in my library](https://github.com/jpmeijers/RN2483-Arduino-Library/blob/master/src/rn2483.cpp#L135), or the gateway close to you does not do downlinks as it is a single channel gateway.

When I'm in the proximity of a gateway I will test to see if I can successfully join using OTAA.

kneibdev, 2016-09-08:
Hi,

Mmm I created a new application, and registered a OTAA device. After running the code on my Sodaq (again I have the same message) I see that the device is activated but noting when listening on MQTT ( 70B3D57ED0000D48/devices/0004A30B001B0F5A/activations )

The Gateway is a Lorank 8. It used to work with croft ...

Thanks!
Oscar

jpmeijers, 2016-12-15:
I have recently used OTAA with this library and it worked successfully. Therefore I mark this as resolved.

closed 2016-12-15 by jpmeijers