jpmeijers/RN2483-Arduino-Library #2: Downlink messages not handled

jpmeijers, 2016-09-08:
This library does not yet handle downlink messages received after transmitting a packet. The correct swtich case is already in place in [rn2483.cpp:286](https://github.com/jpmeijers/RN2483-Arduino-Library/blob/master/src/rn2483.cpp#L286). 

TODO: 
The tx functions should return a struct containing

1. a boolean for confirmed message acks
2. a boolean for received data
3. a string/char array for the downlink/received data

WORKAROUND:
For now a simple Serial.println() can be used in the above mentioned code path to print the received data to the serial port. Alternatively a global variable can be set, which can be checked from the main program.

majid701, 2016-12-12:
@jpmeijers when are you going to implement downlink messages feature? I just checked the code path you mentioned (rnxx3.cpp:286) but that line is out of any function.
Where do i have to put the println function in order to print out downlink messages? 
thanks

jpmeijers, 2016-12-14:
https://github.com/jpmeijers/RN2483-Arduino-Library/blob/master/src/rn2xx3.cpp#L341

I don't expect to have time to do this soon. Not really priority.

Pull requests are welcome though!

majid701, 2016-12-14:
@jpmeijers Ok, i will try to implement it and then send a pull request. Do you have any suggestions for a part from the comments in the specific section of the code (which i read).

thanks

jpmeijers, 2016-12-15:
My original comment said to return a struct. That is nice because the struct can have a field for the received data.

Returning a enum, and adding a separate function to get the received data is the alternative way.

I'm not experienced enough to know what is the best. Let's see how your implementation works and go from there.

majid701, 2016-12-15:
@jpmeijers I was trying to implement the struct but I just noticed it can get confusing for the users and also because all your tx functions return a boolean and so I had to change all of them. So I'm going for the second solution which is just add a function that returns the last downlink message. As we know the downlink should be used as a limited feature this approach allows to keep the last downlink message in memory and retreive it at any moment.

dylanmurray95, 2016-12-19:
@jpmeijers How exactly do you implement the Serial.println() to show the confirmed status sorry? It appears that when I am doing a .txCnf it merely times out as opposed to continuing when confirmed.

dylanmurray95, 2016-12-21:
@jpmeijers I was just wondering, if downlink is not currently supported, does that mean that confirmed messages will not work?

DenniZr, 2017-01-16:
I added downlink functionality. https://github.com/Kaasfabriek/RN2483-Arduino-Library This doesn't mean you have confirmed messenges. Just that you can send something back from the gateway.

jpmeijers, 2017-01-17:
I tested @DenniZr's code with confirmed messages and everything seems to work fine now. The current way the library handles downlinks and acks is by returning a enum value when calling any of the TX functions.

The possible values are:
```
enum TX_RETURN_TYPE {
  TX_FAIL = 0, //The transmission failed. In the case of a confirmed message that is not acked, this will be the returned value.
  TX_SUCCESS = 1, //The transmission was successful. Also the case when a confirmed message was acked.
  TX_WITH_RX = 2 //A downlink message was received after the transmission. This also implies that a confirmed message is acked.
};
```

closed 2017-01-17 by jpmeijers