jpmeijers/RN2483-Arduino-Library #4: Add support for RN2903 module

jpmeijers, 2016-09-19:
The library can currently be slightly modified to work with an RN2903 module. It would be nicer if the module version can be auto detected and the changes applied during run time. 

To get the module version, the command `sys get ver` is sent to the module.
If the reply is `RN2903 0.9.5 Sep 02 2015 17:19:55` it is an RN2903.
The start of this string can be used to apply the correct switch in the software.

The version check should be done in the init() function.

guillermomo, 2016-09-22:
Hi jpmeijers,
                      I want to help you because I need to use the RN2903. Witch kind of info should I need to change.

jpmeijers, 2016-09-23:
I copy a discussion from the TTN slack about this topic:

> telkamp [10:49 PM]  > RN2903 is US, not AU.
> telkamp [10:50 PM]  > You can’t change any of the frequencies, they’re all defined in the LoRaWAN standard.
> 
> bje [12:33 AM]  > @telkamp If RN2903 is not AU, then what is?
> 
> telkamp [8:46 AM]  > @bje let me correct: the RN2903 also is for AU, but the current firmware does not seem to support the LoRaWAN channel plan for AU yet.

@guillermomo Are you using the RN2903 in Australia or the USA?

guillermomo, 2016-09-23:
Hi I am in Argentina, but we used the same frequencies than USA. I check the documentation and I need to change the frequencies to mac set rx2 10 923300000. Thanks.

jpmeijers, 2016-10-21:
We have initial RN2483/RN2903 autodetection in branch: https://github.com/jpmeijers/RN2483-Arduino-Library/tree/RN2xx9-autodetection
Thanks to @nicoschtein

jpmeijers, 2016-10-31:
It's now merged into master.

closed 2016-10-31 by jpmeijers

spabon, 2017-02-16:
Good afternoon dear friends
I need you Help
I have two couples of RN2903  1629CVP modules.
I want to connect them and transmit. Do you have the code to send and receive ? My modules have a frequency of 915 MHz.
Do you have a diagram to connect them?

I thank you, my friend.

guillermomo, 2017-02-17:
Hi Salvador,
                    Can you send me your email and I will send the all information. You can check what I did in this place : https://www.youtube.com/watch?v=QvmbSAuqHJ8. I use the componen RN2903 with the TTN Enschede breakout board.

spabon, 2017-02-17:
Good Morning my firend

This is my email

I need the code for ti implement in ARDUINO the sender and the receiver.

but I have not the diagram.

Thanks.. for you hep

Salvador

[https://i.ytimg.com/vi/QvmbSAuqHJ8/hqdefault.jpg]<https://www.youtube.com/watch?v=QvmbSAuqHJ8>

Welding component RN2903<https://www.youtube.com/watch?v=QvmbSAuqHJ8>
www.youtube.com
IoT

guillermomo, 2017-02-17:
I can see your email please send me your email to willy.moral@gmail.com

spabon, 2017-02-17:
Buenas tardes apreciado profesor

Profe mire es que estoy usando dos módulos para realizar un proyecto y necesito trasmitir con esta tecnología pero no he podido encontrar el código para enviar y recibir.
Lo pienso implementar en Arduino.
Me podría ayudar con estos dos  códigos y las librerías necesarias??

Salvador Pabón Orozco

________________________________
De: salvatore pabonee <sspabonee@hotmail.com>
Enviado: viernes, 17 de febrero de 2017 11:00 a. m.
Para: jpmeijers/RN2483-Arduino-Library
Asunto: Re: [jpmeijers/RN2483-Arduino-Library] Add support for RN2903 module (#4)

Good Morning my firend

This is my email

I need the code for ti implement in ARDUINO the sender and the receiver.

but I have not the diagram.

Thanks.. for you hep

Salvador

[https://i.ytimg.com/vi/QvmbSAuqHJ8/hqdefault.jpg]<https://www.youtube.com/watch?v=QvmbSAuqHJ8>