jpmeijers/RN2483-Arduino-Library #37: modifications for single-channel gateways

eliabieri, 2017-02-14:
Hello everyone,
what modifications would I have to do to use this library with a single channel gateway?
I thought about changing the duty cycles in a way, that only one channel is used in reality.
I would do this in this section of code: 
''

        //RX window 2
        sendRawCommand(F("mac set rx2 3 869525000"));

        //channel 0
        sendRawCommand(F("mac set ch dcycle 0 799"));

        //channel 1
        sendRawCommand(F("mac set ch drrange 1 0 6"));
        sendRawCommand(F("mac set ch dcycle 1 799"));

        //channel 2
        sendRawCommand(F("mac set ch dcycle 2 799"));

        //channel 3
        sendRawCommand(F("mac set ch freq 3 867100000"));
        sendRawCommand(F("mac set ch drrange 3 0 5"));
        sendRawCommand(F("mac set ch dcycle 3 799"));
        sendRawCommand(F("mac set ch status 3 on"));

        //channel 4
        sendRawCommand(F("mac set ch freq 4 867300000"));
        sendRawCommand(F("mac set ch drrange 4 0 5"));
        sendRawCommand(F("mac set ch dcycle 4 799"));
        sendRawCommand(F("mac set ch status 4 on"));

        //channel 5
        sendRawCommand(F("mac set ch freq 5 867500000"));
        sendRawCommand(F("mac set ch drrange 5 0 5"));
        sendRawCommand(F("mac set ch dcycle 5 799"));
        sendRawCommand(F("mac set ch status 5 on"));

        //channel 6
        sendRawCommand(F("mac set ch freq 6 867700000"));
        sendRawCommand(F("mac set ch drrange 6 0 5"));
        sendRawCommand(F("mac set ch dcycle 6 799"));
        sendRawCommand(F("mac set ch status 6 on"));

        //channel 7
        sendRawCommand(F("mac set ch freq 7 867900000"));
        sendRawCommand(F("mac set ch drrange 7 0 5"));
        sendRawCommand(F("mac set ch dcycle 7 799"));
        sendRawCommand(F("mac set ch status 7 on"));

        returnValue = true;
      }
      else
      {
        returnValue = false;
      }

      break;
    }

    case TTN_US:
    {
    /*
     * Most of the TTN_US frequency plan was copied from:
     * https://github.com/TheThingsNetwork/arduino-device-lib
     */
      if(_moduleType == RN2903)
      {
        for(int channel=0; channel<72; channel++)
        {
          // Build command string. First init, then add int.
          String command = F("mac set ch status ");
          command += channel;

          if(channel>=8 && channel<16)
          {
            sendRawCommand(command+F(" on"));
          }
          else
          {
            sendRawCommand(command+F(" off"));
          }
        }
        returnValue = true;
      }
      else
      {
        returnValue = false;
      }
      break;
    }

    case DEFAULT_EU:
    {
      if(_moduleType == RN2483)
      {
        //fix duty cycle - 1% = 0.33% per channel
        sendRawCommand(F("mac set ch dcycle 0 799"));
        sendRawCommand(F("mac set ch dcycle 1 799"));
        sendRawCommand(F("mac set ch dcycle 2 799"));

        //disable non-default channels
        sendRawCommand(F("mac set ch status 3 on"));
        sendRawCommand(F("mac set ch status 4 on"));
        sendRawCommand(F("mac set ch status 5 on"));
        sendRawCommand(F("mac set ch status 6 on"));
        sendRawCommand(F("mac set ch status 7 on"));

        returnValue = true;
      }
      else
      {
        returnValue = false;
      }

      break;
    }
    default:
    {
      //set default channels 868.1, 868.3 and 868.5?
      returnValue = false; //well we didn't do anything, so yes, false
      break;

''
Am I right, and how would I do it exactly?
I use [this](https://github.com/JaapBraam/LoRaWanGateway) single channel library and I listen on the following frequency: 868100000

jpmeijers, 2017-02-14:
All you have to do is call:
`myLora.setFrequencyPlan(SINGLE_CHANNEL_EU);`
after myLora.init.

closed 2017-02-14 by jpmeijers

eliabieri, 2017-02-14:
Thank you very much. Thats all I wanted to know.

eliabieri, 2017-02-14:
Ok thanks for the help.
Can you explain me the exact difference between these two calls?
`     sendRawCommand(F("mac set rx2 5 868100000")); //use this for "strict" one channel gateways
        sendRawCommand(F("mac set rx2 3 869525000")); //use for "non-strict" one channel gateways`

jpmeijers, 2017-02-14:
That is only for when you use this code:
https://github.com/things4u/ESP-1ch-Gateway-v3.0

Their documentation should say what the differences are.

In general we do not like "strict" single channel gateways as they are even further away from the lorawan specification.