jpmeijers/RN2483-Arduino-Library #43: Select sub band

sabas1080, 2017-03-14:
Hi @jpmeijers 

I'm trying to connect to sub band 7

FSB 7 (aka block G) = 48, 49, 50, 51, 52, 53, 54, 55 plus 70

Do you have any examples of how to do this?
 
I am manually using the RN2903 and it works correctly

'mac set dr 3'
'mac set adr off'
'mac set ar off'
'mac set sync 34'

Thanks

jpmeijers, 2017-03-15:
Hi @sabas1080 

I can add a frequency plan for your area. 

In what country/area are you?
Are you using The Things Network?
Could you perhaps share your full working code so that I can see all configuration you do with the radio?

sabas1080, 2017-03-15:
I have using TTN

My country is Mexico, 915 Mhz

I use a python script for testing RN2903 and work

https://github.com/sabas1080/ttn_node_py/blob/master/lora_mote_send_otaa.py

I use suband 7 with a multitech gateway

Thanks

closed 2025-09-01 by sabas1080