jpmeijers/RN2483-Arduino-Library #64: About ESP32+RN2483 LoRa module

uceekji, 2019-08-01:
I have used ESP32 and RN2483 for my lora project. When I changed the PIN set of my ESP32, I upload the program 'ESP9266-RN2483-basic.ino'. But it said that connection unsuccessful and power cycle on the board. I don't know whats the problem but I can make sure the hardware system including connection is well.

jpmeijers, 2019-08-01:
A good way to start is to use a SerialPassthrough sketch on the ESP32. Then you can try and send commands directly to the module and see if it responds. Something like `sys get ver`. When you reset the RN2483 is should also print out the version number directly after it startup up again. Are you using 56700 baud?

uceekji, 2019-08-01:
Yes, it is 57600 baud. What is the meaning of SerialPassthrough sketch.

Thank you.

uceekji, 2019-08-04:
I also want to know, how to repeat to start up this program instead of reset ESP32. Because sometimes I only want to reset the program not the MCU.

Thank you.